reclosedev/pyautocad #8: Incompatible with python 3.x

S0mbre, 2015-12-21:
The library's source files contain syntax incompatible with python 3.x, notably the print statement (when it should be a function). The syntax is otherwise not yet adapted for python 3.x, e.g.: old-style string formatting (the % operator), old-style exception handlers (comma instead of 'as'), and so on. The total amount of corrections to be made for python 3.x support is not so large.

As of version 0.1.3, pyautocad causes syntax errors on python 3.4. Please update the library ASAP.

reclosedev, 2015-12-21:
I've added Python 3 compatibility. You could try version 0.2.0, it's released on PyPI.

Unit tests are passed. Only test_contrib_tables.py fails due to ``tablib`` incompatibility. 
I don't use AutoCAD (or any CAD) anymore, and can't fully test it.

JuanJ67m, 2020-09-10:
```

>>> from pyautocad import Autocad, APoint
>>> acad = Autocad()
>>> p0 = APoint(0,0)
>>> p1 = APoint(0,2)
>>> p0
APoint(0.00, 0.00, 0.00)
>>> p1
APoint(0.00, 2.00, 0.00)
>>> acad.model.AddEllipse(p0, p1, 0.5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python38\lib\site-packages\pyautocad\api.py", line 85, in model
    return self.doc.ModelSpace
  File "C:\Python38\lib\site-packages\pyautocad\api.py", line 74, in doc
    return self.app.ActiveDocument
  File "C:\Python38\lib\site-packages\pyautocad\api.py", line 63, in app
    self._app = comtypes.client.GetActiveObject('AutoCAD.Application', dynamic=True)
  File "C:\Python38\lib\site-packages\comtypes\client\__init__.py", line 180, in GetActiveObject
    obj = comtypes.GetActiveObject(clsid, interface=interface)
  File "C:\Python38\lib\site-packages\comtypes\__init__.py", line 1245, in GetActiveObject
    oledll.oleaut32.GetActiveObject(byref(clsid), None, byref(p))
  File "_ctypes/callproc.c", line 948, in GetResult
OSError: [WinError -2147221021] Operation unavailable
>>>
```