reclosedev/pyautocad #13: Cann't open autocad

qingsonger, 2016-11-05:
I run the example, but cann't do it!
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    acad.prompt("Hello, Autocad from Python\n")
  File "C:\Program Files\Miniconda2\lib\site-packages\pyautocad\api.py", line 162, in prompt
    self.doc.Utility.Prompt(u"%s\n" % text)
  File "C:\Program Files\Miniconda2\lib\site-packages\pyautocad\api.py", line 74, in doc
    return self.app.ActiveDocument
  File "C:\Program Files\Miniconda2\lib\site-packages\pyautocad\api.py", line 63, in app
    self._app = comtypes.client.GetActiveObject('AutoCAD.Application', dynamic=True)
  File "C:\Program Files\Miniconda2\lib\site-packages\comtypes\client\__init__.py", line 173, in GetActiveObject
    clsid = comtypes.GUID.from_progid(progid)
  File "C:\Program Files\Miniconda2\lib\site-packages\comtypes\GUID.py", line 78, in from_progid
    _CLSIDFromProgID(unicode(progid), byref(inst))
  File "_ctypes/callproc.c", line 946, in GetResult
WindowsError: [Error -2147221005]
I use win7, python 2.7 64bit. Autocad2014

vaish1234567, 2017-11-09:
I am also getting the same problem.Please provide the solution asap

luozn15, 2019-08-25:
same problem too

alvin2ye, 2019-10-22:
same problem too