reclosedev/pyautocad #14: cannot find cad

unmht001, 2017-02-25:
i installed 2 versions of autocad  cad2017 and   cad2014 mechanical  , 
i used code like this:

from pyautocad import Autocad
a=Autocad()
a.prompt("hello")

cad2014 mechanical was opened , why not cad2017?
how to run this in cad2017?

unmht001, 2017-02-25:
i read api.py , i found this:
#-----------------
def __init__(self, create_if_not_exists=False, visible=True):
....
#--------------------
and 
this :
#----------------
def app(self):
....
self._app = comtypes.client.GetActiveObject('AutoCAD.Application', dynamic=True) 
#-------------------

i think , if __init__()  and app() could know the version of autocad , will be better.
like this . 

self._app = comtypes.client.GetActiveObject('AutoCAD.Application.21', dynamic=True) 

and it worked.

vaish1234567, 2017-11-20:
What is .21 in this self._app = comtypes.client.GetActiveObject('AutoCAD.Application.21', dynamic=True)?