reclosedev/pyautocad #17: _ctypes.COMError

YU6326, 2017-10-20:
when type the example:
```python
for obj in acad.iter_objects():
    print obj.ObjectName
```
a bug occurred
Traceback (most recent call last):
  File "<pyshell#10>", line 1, in <module>
    for obj in acad.iter_objects():
  File "C:\Program Files\Python36\lib\site-packages\pyautocad\api.py", line 128, in iter_objects
    item = self.best_interface(item)
  File "C:\Program Files\Python36\lib\site-packages\pyautocad\api.py", line 156, in best_interface
    return comtypes.client.GetBestInterface(obj)
  File "C:\Program Files\Python36\lib\site-packages\comtypes\client\__init__.py", line 126, in GetBestInterface
    result = punk.QueryInterface(interface)
  File "C:\Program Files\Python36\lib\site-packages\comtypes\client\lazybind.py", line 142, in QueryInterface
    return self._comobj.QueryInterface(*args)
  File "C:\Program Files\Python36\lib\site-packages\comtypes\__init__.py", line 1156, in QueryInterface
    self.__com_QueryInterface(byref(iid), byref(p))
_ctypes.COMError: (-2147467262, '不支持此接口', (None, None, None, 0, None))

zacharynevin, 2019-05-22:
I'm getting the same thing. My system specs are:

- macOS Mojave 10.14.5
- Python 3.7.2 running in virtualenv.
- pip 19.1.1.

speilberg0, 2019-12-23:
I worked around this issue by removing old versions of acax*enu.tlb and axdb*enu.tlb from "C:\Program Files\Common Files\Autodesk Shared".

Eg. I'm currently running Autocad 2020, therefore please ensure only acax23enu.tlb and axdb23enu.tlb are available. 

Relevant area of code is api.py lines 22-28