reclosedev/pyautocad #19: Cannot create Autocad object

vaish1234567, 2017-11-20:
I tried the following :
from pyautocad import Autocad, APoint
acad = Autocad(create_if_not_exists=True)
acad.prompt("Hello, Autocad from Python\n")

The following is the error i m getting :
>>> acad.prompt("Hello, Autocad from Python\n")                                
 Hello, Autocad from Python                                                                                                                                      Traceback (most recent call last):                                               
 File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\site-packages\pyautocad\api.py", line 62, in app                                                 
self._app = comtypes.client.GetActiveObject('AutoCAD.Application', dynamic=True)                                                                              File"C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\sitepackages\comtypes\client\__init__.py", line 173, in GetActiveObject                         
clsid = comtypes.GUID.from_progid(progid)                                     
File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\site-packages\comtypes\GUID.py", line 78, in from_progid                                        
 _CLSIDFromProgID(str(progid), byref(inst))                                    
File "_ctypes/callproc.c", line 918, in GetResult                             
OSError: [WinError -2147221005] Invalid class string                                                                                                            During handling of the above exception, another exception occurred:                                                                                             Traceback (most recent call last):                                                
File "<stdin>", line 1, in <module>                                             
File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\site-packages\pyautocad\api.py", line 161, in prompt                                            
 self.doc.Utility.Prompt(u"%s\n" % text)                                       
File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\site-packages\pyautocad\api.py", line 73, in doc                                                
 return self.app.ActiveDocument                                                
File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\site-packages\pyautocad\api.py", line 66, in app                                                
 self._app = comtypes.client.CreateObject('AutoCAD.Application', dynamic=True)                                                                                 File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\sitepackages\comtypes\client\__init__.py", line 227, in CreateObject                           
 clsid = comtypes.GUID.from_progid(progid)                                    
 File "C:\Users\snehal\AppData\Local\Programs\Python\Python36\lib\site-packages\comtypes\GUID.py", line 78, in from_progid                                         
_CLSIDFromProgID(str(progid), byref(inst))                                    
File "_ctypes/callproc.c", line 918, in GetResult                            
 OSError: [WinError -2147221005] Invalid class string                            
>>>                                                                           
I am using Python3.6 ,Autocad 2016,Windows 8 (64bit). Please provide me the solution. Thank you

HuangYongfei, 2018-03-30:
Hi, I also have this problem, Do you solve it?

k-and-d, 2021-04-30:
I solved this error by understanding how com types work 

When we create an Autocad object using 

`acad = Autocad()`

`pyautocad` creates an object using comtypes by calling `comtypes.client.GetActiveObject('AutoCAD.Application', dynamic=True)`

'**AutoCAD.Application**' is a comtype registered in the windows registry 

You should find it under 

HKEY_LOCAL_MACHINE\SOFTWARE\Classes\AutoCAD.Application

however in the our case when the exception **Invalid class string** means that the path doesnt exist

So to fix this you may clone AutoCAD.Application registry from another path 

HKEY_CURRENT_USER\SOFTWARE\Classes\AutoCAD.Application

Cloning all keys under the AutoCAD.Application from HKEY_CURRENT_USER to HKEY_LOCAL_MACHINE fixed it in my machine

I copied the keys by creating new keys manually in the target location 

Just copied one registry key : AutoCAD.Application
```
AutoCAD.Application:{
CLSID {8B4929F8-076F-4AEC-AFEE-8928747B7AE3}
CurVer AutoCAD.Application.24
}
```

Using Autocad 2020 and python 2.7