reclosedev/pyautocad #24: howto polyline?

nortikin, 2019-02-15:
>>> m.AddPolyline((APoint(0,0),APoint(-1000,-1000),APoint(5000,-5000)))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\P1\AppData\Local\Programs\Python\Python36\lib\site-packages\comtypes\client\lazybind.py", line 182, in caller
    return self._comobj._invoke(descr.memid, descr.invkind, 0, *args)
  File "C:\Users\P1\AppData\Local\Programs\Python\Python36\lib\site-packages\comtypes\automation.py", line 729, in _invoke
    dp, var, None, argerr)
_ctypes.COMError: (-2147352567, 'Ошибка.', (None, None, None, 0, None))

BlitterTech, 2019-05-06:
```python
from pyautocad import Autocad
import array

acad = Autocad(create_if_not_exists=True)

points_2d = [0, 0, 1, 1, 2, 1]
points_double = array.array("d", points_2d)
acad.model.AddLightWeightPolyline(points_double)

points_3d = [0, 0, 0, 1, 1, 0, 2, 1, 0]
points_double = array.array("d", points_3d)
acad.model.AddPolyline(points_double)
```

iNakel, 2021-09-14:
> ```python
> from pyautocad import Autocad
> import array
>
> acad = Autocad(create_if_not_exists=True)
>
> points_2d = [0, 0, 1, 1, 2, 1]
> points_double = array.array("d", points_2d) [...]

the result is a 2D Polyline... and not a 3D Polyline... ¿is it possible to get it 3D?

better319, 2023-09-01:
in the end how to choice the “closed”

better319, 2023-09-01:
in the end how to choice the “closed”，when finish draw

CEXT-Dan, 2023-09-01:
> in the end how to choice the “closed”，when finish draw

use Closed..
https://help.autodesk.com/view/OARX/2023/ENU/?guid=GUID-5A604B98-B6DE-42C3-97D9-98B8CAEBB507

`from pyautocad import Autocad
import array

acad = Autocad(create_if_not_exists=True)

points_3d = [0, 0, 0, 1, 1, 0, 2, 1, 0]
points_double = array.array("d", points_3d)
pline = acad.model.AddPolyline(points_double)
pline.Closed = True`