reclosedev/pyautocad #34: MacOS

larafdouglas, 2021-04-16:
Is possible to run pyautocad in a mac?
Im trying and I get this error: 

`
Python 3.6.8 (v3.6.8:3c6b436a57, Dec 24 2018, 02:04:31) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyautocad
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyautocad/__init__.py", line 15, in <module>
    from pyautocad.api import *
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/pyautocad/api.py", line 16, in <module>
    import comtypes
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/comtypes/__init__.py", line 23, in <module>
    from _ctypes import COMError
ImportError: cannot import name 'COMError'`

hotarumoon, 2021-09-13:
I am having the exact same error, I have tried with different versions of Python as well

CWojahn, 2022-11-30:
Same problem here

q619538185mm, 2023-07-23:
I have the same problem, and I tried to change different cad versions from 2020 to 2023 as well.