reclosedev/pyautocad #48: Open DWG files and manipulate them

Bregolas3, 2023-10-29:
I am trying to figure out if it is possible to open multiple already existing dwg files through pyautocad and manipulate them to create a single dwg file. Basically, I am trying to figure out if I can take various drawings of different length walls and manipulate those walls to create a single room. I am trying to automate this process as each wall is already drawn and it is tedious to pull in each wall every time I go to draw a room. Does pyautocad support something like this? I was unable to find anything in the documentation on opening and manipulating previously created DWG files.

CEXT-Dan, 2023-10-29:
Sure, you need a CAD platform like AutoCAD, 
with pyautocad you can use AutoCAD's ActiveX API to insert each sub drawing into a main drawing with 
InsertBlock, you can then manipulate each item in the block, or explode all the blocks into modelspace

Bregolas3, 2023-10-29:
Would you mind possibly walking me through how I could take four walls and manipulate them in a way where I rotate the walls to create a room in this capacity? If I have each wall drawn and I just need to insert them into the drawing, I would need to make each wall a block right? And then pyautocad lets me rotate each wall and place it in a manner where they line up so it creates the room right? Would a previously existing subdrawing be able to be placed in this manner?

CEXT-Dan, 2023-10-29:
Inserting blocks is a fundamental operation you should probably understand before you begin writing code.

it’s also a good idea to become somewhat familiar with the API pyautocad accesses 
https://help.autodesk.com/view/OARX/2023/ENU/?guid=GUID-A809CD71-4655-44E2-B674-1FE200B9FE30

insertBlock imports the drawing and creates the block(s) for you
The insert block has insertion point, scale and rotation arguments
https://help.autodesk.com/view/OARX/2023/ENU/?guid=GUID-17F86FDD-B7FC-4F43-9F16-B4958F73A66D

`pseudo code 
from pyautocad import Autocad,APoint
acad = Autocad(create_if_not_exists = True)

#one for each wall
point = APoint(0,0)
wallX = acad.model.InsertBlock(point, “wallpath.dwg”, scaleX, ScaleY, scaleZ, Rotation)

doc = acad.ActiveDocument

# Save
doc.SaveAs("C: \\room.dwg")

`