reclosedev/pyautocad #51: Running functions that takes in symbol as input

JonathanHandojo, 2023-12-18:
I'm just wondering, how can you pass on symbols as arguments here? For example, I very much need to use the GetBoundingBox and GetEntity function, but I can't seem to get this working because both function need to take in a symbol.

[image]

Or if this is not possible, then what's the other alternative to solve this?

CEXT-Dan, 2023-12-18:
the function should return a tuple with the min max values. 
I think you call the entity.GetBoundingBox() with no arguments

JonathanHandojo, 2023-12-18:
Yea, I tried that, it returned an error of **_ctypes.COMError: (-2147352562, 'Invalid number of parameters.', (None, None, None, 0, None))_**

CEXT-Dan, 2023-12-18:
Yeah, I couldn’t get it to work. 
As part of my project, https://github.com/CEXT-Dan/PyRx
See https://github.com/CEXT-Dan/PyRx/blob/main/PyRxStubs/AxApp24.py

I used win32com’s makepy to generate wrappers, and that works.
I modified the generated wrappers in that I added in a sequence to variant, as needed.

If you can’t use my project, maybe it will give you inspiration 

`
def PyRxCmd_GetEntX():
    try:
        doc: Ax.IAcadDocument = theApp.ActiveDocument
        util: Ax.IAcadUtility = doc.Utility
        result = util.GetEntity("\nPick an ent: ")
        ent: Ax.IAcadEntity = result[0]
        print(ent.ObjectName, ent)
        print(ent.GetBoundingBox(), ent)
    except Exception as err:
        traceback.print_exception(err)
`

`
Command: AcDbLine <win32com.gen_py.AutoCAD 2021 Type Library.IAcadLine instance at 0x2601015985504>
((15.59022968248874, 6.720747514960074, 0.0), (26.48143751382426, 13.071158956573747, 0.0)) <win32com.gen_py.AutoCAD 2021 Type Library.IAcadLine instance at 0x2601015985504>
`

JonathanHandojo, 2023-12-19:
Wow... I haven't tried it yet, but just by looking at it, that's a very impressive project. I'll follow the steps to install this and give it a shot. I'll let you know how I go. Thanks!

JonathanHandojo, 2023-12-19:
> Yeah, I couldn’t get it to work. As part of my project, https://github.com/CEXT-Dan/PyRx See https://github.com/CEXT-Dan/PyRx/blob/main/PyRxStubs/AxApp24.py
>
> I used win32com’s makepy to generate [...]

Hi Dan,

I tried installing PyRx, but I can't seem to import its libraries. Upon running the installation, it says 'Python not found', even though I have it installed.

CEXT-Dan, 2023-12-19:
If the installer is complaining, you can double check that python is in your path(s)
Before running the installer, make sure the python has a path

https://github.com/CEXT-Dan/PyRx/blob/main/env.png

CEXT-Dan, 2023-12-19:
BTW, I’ll be out of of touch  for the holidays, I fly out in the morning.. kind of bad timing

There’s a ton of info at the swamp that may help you.. kind of a history of my progress and a bunch of samples
https://www.theswamp.org/index.php?board=76.0
https://www.theswamp.org/index.php?topic=58162.0

JonathanHandojo, 2023-12-19:
Hi, no worries, have a good holiday.

I finally got the modules loaded. However, I wasn't sure from which theApp was referring to.

I made it as such, but still yields an error:

```
from pyautocad import Autocad
import AxApp24 as Ax

def PyRxCmd_GetEntX():

    acad = Autocad()
    try:
        doc: Ax.IAcadDocument = acad.ActiveDocument
        util: Ax.IAcadUtility = doc.Utility
        result = util.GetEntity("\nPick an ent: ")
        ent: Ax.IAcadEntity = result[0]
        print(ent.ObjectName, ent)
        print(ent.GetBoundingBox(), ent)
    except Exception as err:
        print(err)

PyRxCmd_GetEntX()
```

Unless I'm misunderstanding something... but I'll continue to investigate and try again.

CEXT-Dan, 2023-12-19:
Great! you can just get rid of the pyautocad import and stuff

You can try my test
https://github.com/CEXT-Dan/PyRx/blob/main/PySamples/ActiveX/testWin32Com.py

in AutoCAD, use the new pyload command, load  testWin32Com.py
and type GetEntX()

let me know how that works.