reclosedev/pyautocad #55: AttributeError: ActiveDocument

Ammatarasu, 2024-04-24:
I'm trying to set up a small script to run a lisp routine over a collection of .dwg files. This is part of a bigger script to export ArcGIS featureclass files into specially formatted .dwg files. The export to CAD function in Arcgis doesn't work well with MLines so I have to run a secondary pass over the generated files to convert the Polylines into MLines.

When running this code:
```
AutoCadFiles = [os.path.join(exportFolder,file) for file in os.listdir(exportFolder) if file.endswith('.dwg')]
acad = pyautocad.Autocad()
for file in AutoCadFiles:
    acad.app.Documents.Open(file)
    acad.doc.PostCommand("._zoom _e ._PLtoML N ")
    acad.doc.Close('True')
```
I sometimes get an error :
```
AttributeError                            Traceback (most recent call last)
In  [15]:
Line 14:    acad.doc.PostCommand("._zoom _e ._PLtoML N ")

File C:\Users\daandevries\AppData\Roaming\Python\Python39\site-packages\pyautocad\api.py, in doc:
Line 74:    return self.app.ActiveDocument

File C:\Users\daandevries\AppData\Roaming\Python\Python39\site-packages\comtypes\client\lazybind.py, in __getattr__:
Line 147:   raise AttributeError(name)

AttributeError: ActiveDocument
-----------------------------------
```

It seems to me like the PostCommand function might be trying to execute before the opened document has finished opening. I saw in the AutoCAD ActiveX documentation that AutoCAD does have an EndOpen event ([[https://help.autodesk.com/view/OARX/2023/ENU/?guid=GUID-1AEC7F00-981A-4517-A830-37CA653D336A]])
Except I have no idea how i could intergrate this in my own code. Can you help me with this?

This error does always occur eventually. Sometimes it does do the first file correctly.
I also tried it with just the ```acad.doc.PostCommand("._zoom _e ._PLtoML N Qsave Close"). Except this can also run into the same issues.

CEXT-Dan, 2024-04-24:
Documents.Open returns a document, what if you send the command with that instance?

```Python
        axDoc = axApp.Documents.Open(file)
        axDoc.SendCommand(...)
        axDoc.Close(...)
```

Ammatarasu, 2024-04-24:
I found that added 2 time.sleep() statements in between when the doc opens and the PostCommand section.
I was able to remove 1 of those by using the document instance as you described below. Unfortunately i seem to be getting a COMError now when i execute this. The first document goes fine but opening the second doc doesnt work.
```
for file in AutoCadFiles:
    axDoc = acad.app.Documents.Open(file)
    axDoc.PostCommand("._zoom _e ._Revisie n Qsave Close ")
```
```axDoc.Close('True') also gives me a COM error, so I've gone back to using the PostCommand to close the file again.

This is the exception traceback I'm getting:
```
---------------------------------------------------------------------------
COMError                                  Traceback (most recent call last)
In  [14]:
Line 27:    axDoc = acad.app.Documents.Open(file)

File C:\Users\daandevries\AppData\Roaming\Python\Python39\site-packages\comtypes\client\dynamic.py, in __getattr__:
Line 120:   dispid = self._comobj.GetIDsOfNames(name)[0]

File C:\Users\daandevries\AppData\Roaming\Python\Python39\site-packages\comtypes\automation.py, in GetIDsOfNames:
Line 823:   self.__com_GetIDsOfNames(riid_null, arr, len(names), lcid, ids)

COMError: (-2147418111, 'Aanroep geweigerd door aangeroepene.', (None, None, None, 0, None))
---------------------------------------------------------------------------
```
Translated ("Call rejected by callee")

Giving it 2 to 3 seconds after the PostCommand seems to work most of the time on my pc. However this is probably on a case by case basis depending on the file size and the performance of the PC thats running the script.

CEXT-Dan, 2024-04-24:
maybe you should set the document that you have opened to the 'ActiveDocument'

```Python
        axApp = Ax.getApp()
        axApp.ActiveDocument =  axApp.Documents.Open('E:/Temp.dwg')
        axApp.ActiveDocument.PostCommand(...)
        axApp.ZoomExtents()
        axApp.ActiveDocument.Close()
```

CEXT-Dan, 2024-04-25:
Ah sorry, getapp() is from the project I’m working on
https://github.com/CEXT-Dan/PyRx/blob/0715a3fb5d37b73e4e8dc3c207abcc882453b198/PyRxStubs/AxApp24.py#L32954-L32963
if your able to run these modules you might have better results since python is loaded into AutoCAD’s process. It may not work for what you’re doing though 

I don’t know how to use the events from PyAutoCAD, you might get hints from the link above (see createEventObject)
And https://github.com/CEXT-Dan/PyRx/blob/main/PySamples/ActiveX/testEvents.py