reclosedev/pyautocad #57: COM Error -2147352567 -2145320944 autocad new version

pyautocad, 2024-06-09:
Hello everyone,

I erase an old version of autocad and downloaded the 2025 one. Many errors are displayed:

I tested some simple examples:

\>>> from pyautocad import Autocad, APoint

\>>> acad = Autocad(create_if_not_exists=True)

\>>> acad.prompt("Hello, Autocad from Python\n")

Hello, Autocad from Python

this works but for this one, it doesn't

>>> import pyautocad
>>>
>>> acad = pyautocad.Autocad()
>>>
>>> center_point = pyautocad.aDouble(5, 5)  # ← ← ←
>>>
>>> radius = 3
>>>
>>> circle = acad.model.AddCircle(center_point, radius)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\allou\AppData\Local\Programs\Python\Python312\Lib\site-packages\comtypes\client\lazybind.py", line 166, in caller
    return self._comobj._invoke(descr.memid, descr.invkind, 0, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\allou\AppData\Local\Programs\Python\Python312\Lib\site-packages\comtypes\automation.py", line 843, in _invoke
    self.__com_Invoke(memid, riid_null, lcid, invkind, dp, var, None, argerr)
_ctypes.COMError: (-2147352567, 'Une exception s’est produite.', ("Le nombre d'éléments dans SafeArray est incorrect", 'AutoCAD.Application', 'C:\\Program Files\\Autodesk\\AutoCAD 2025\\HELP\\OLE_ERR.CHM', -2145320944, None))
>>>
>>> print("Circle drawn successfully.")
Circle drawn successfully

        
Do you have any idea please?

CEXT-Dan, 2024-06-09:
center_point  must have x, y, z

pyautocad, 2024-06-09:
Thank you CEXT-Dan! You are right, it works now. Actually, I was working on a bigger program and this was just a little check, this error appears even with adding a third coordinate. You can have a look on it if you want (this is the second part of the program, the first part works well, it's just collecting some datas on excel). The im is to build a map with some coordinates and to draw squares, the squares will be filled of some colours associated by an another value qc.                                                                   Thnk you again for your kind help :)                                                                                                                                >>> # Definition of mean value ranges and corresponding colors
>>> color_ranges = [
...     (1, 1.5, 255),   # White for values between 1 and 1.5
...     (1.5, 2, 3),     # Green for values between 1.5 and 2
...     (2, 2.5, 5),     # Yellow for values between 2 and 2.5
...     (2.5, 3, 1)      # Red for values between 2.5 and 3
...     # Add other value ranges and colors as needed
... ]
>>>
>>> # Function to get the color index based on the mean value
>>> def get_color(mean_qc):
...     for min_range, max_range, color_index in color_ranges:
...         if min_range <= mean_qc < max_range:
...             return color_index  # Return the corresponding color index
...     return 0  # Default color index (black)
...
>>> # Function to create and save the map
>>> def create_map(coordinates_and_means, all_files_coordinates, title, output_path, depth_min, depth_max):
...     print("\nCreating the map...")
...     try:
...         print("Initializing AutoCAD...")
...         acad = Autocad(create_if_not_exists=True)
...         doc = acad.doc
...         model_space = doc.ModelSpace
...         print("AutoCAD initialized. Setting up the map...")
...         square_size = 100  # Size of the squares
...         text_height = 50   # Height of the text
...         padding = 100      # Spacing between elements
...         print("Calculating map boundaries...")
...         e_coords = [coord[1] for coord in coordinates_and_means]
...         n_coords = [coord[2] for coord in coordinates_and_means]
...         min_e, max_e = min(e_coords), max(e_coords)
...         min_n, max_n = min(n_coords), max(n_coords)
...         width = max_e - min_e
...         height = max_n - min_n
...         print("Adding included files...")
...         for filename, e, n, mean_qc in coordinates_and_means:
...             print(f"Processing file: {filename}, e: {e}, n: {n}, mean_qc: {mean_qc}")
...             color_index = get_color(mean_qc)
...             print(f"Color index: {color_index}")
...             insert_point = pyautocad.aDouble(e, n, 0)  # Adding the z coordinate
...             square_points = [
...                 pyautocad.aDouble(e - square_size/2, n + square_size/2, 0),
...                 pyautocad.aDouble(e + square_size/2, n + square_size/2, 0),
...                 pyautocad.aDouble(e + square_size/2, n - square_size/2, 0),
...                 pyautocad.aDouble(e - square_size/2, n - square_size/2, 0),
...                 pyautocad.aDouble(e - square_size/2, n + square_size/2, 0)  # Close the polyline
...             ]
...             polyline = model_space.AddPolyline(square_points)
...             polyline.TrueColor = color_index
...             model_space.AddText(filename, insert_point, text_height)
...         print("Adding non-included files...")
...         for filename, e, n in all_files_coordinates:
...             if not any(filename == coord[0] and e == coord[1] and n == coord[2] for coord in coordinates_and_means):
...                 print(f"Processing non-included file: {filename}, e: {e}, n: {n}")
...                 insert_point = pyautocad.aDouble(e, n, 0)  # Adding the z coordinate
...                 square_points = [
...                     pyautocad.aDouble(e - square_size/2, n + square_size/2, 0),
...                     pyautocad.aDouble(e + square_size/2, n + square_size/2, 0),
...                     pyautocad.aDouble(e + square_size/2, n - square_size/2, 0),
...                     pyautocad.aDouble(e - square_size/2, n - square_size/2, 0),
...                     pyautocad.aDouble(e - square_size/2, n + square_size/2, 0)  # Close the polyline
...                 ]
...                 polyline = model_space.AddPolyline(square_points)
...                 polyline.TrueColor = 128
...                 model_space.AddText(filename, insert_point, text_height)
...         print("Setting up the view...")
...         doc.ActiveView.Fit(True)
...         doc.ActiveView.Height = height + padding
...         doc.ActiveView.Width = width + padding
[... 20 lines cut ...]
>>>
>>> # Create and save the map for the upper zone
>>> create_map(coordinates_and_means_upper, all_files_coordinates_upper, 'Mean cone resistance, mean (qc) for the upper weak zone [MPa]', upper_map_path, 1, 3)

Creating the map...
Initializing AutoCAD...
AutoCAD initialized. Setting up the map...
Calculating map boundaries...
Adding included files...
Processing file: file1, e: 10, n: 20, mean_qc: 1.2
Color index: 255
An error occurred while creating the map: (-2147352567, 'Une exception s’est produite.', (None, None, None, 0, None))
>>>
>>> # Create and save the map for the lower zone
>>> create_map(coordinates_and_means_lower, all_files_coordinates_lower, 'Mean cone resistance, mean (qc) for the lower weak zone [MPa]', lower_map_path, 4.5, 6.5)

Creating the map...
Initializing AutoCAD...
AutoCAD initialized. Setting up the map...
Calculating map boundaries...
Adding included files...
Processing file: file5, e: 50, n: 60, mean_qc: 3.5
Color index: 0
An error occurred while creating the map: (-2147352567, 'Une exception s’est produite.', (None, None, None, 0, None))
>>>